hz-b/rayx #62: Our simple Toroid test fails

treegardel, 2022-06-28:
Test https://github.com/hz-b/RAY-X/blob/master/Tests/rml_files/test_shader/toroid.rml fails:

At the image plane we should see the following foodprint:
[image]

But we see (right side):
[image]

treegardel, 2022-07-04:
@OSMaxwell
I was able to get the debugger running with your help and with the debugger I could fix the bug:
[image]

So that we can turn on the test again.

memoryleak47, 2022-07-29:
Currently, on development, Ray X and Ray UI yield different rays for this test. So to me it doesn't seem fixed yet.
Maybe something not yet merged?

OSMaxwell, 2022-08-12:
Yes it's still seems wrong in the scaling Y and X intervals..
[image]

memoryleak47, 2022-08-21:
[image]
For me it looks like this.

memoryleak47, 2022-08-21:
Yes, good observation!
The lowest y position is > -0.12 for RAYX and < -0.14 in RAY-UI.
The reason for this seems to be that a few rays in RAY-X miss the toroid.

As one can see in https://github.com/hz-b/RAY-X/issues/62#issuecomment-1173943272 above.

memoryleak47, 2022-08-21:
It seems to be a wasteBox-bug of the toroid.
the wasteBox excludes too many rays.

If I return 1.0 in wasteBox all the time the test actually passes.

memoryleak47, 2022-08-21:
Lustig, das Problem war, dass die Argumente zLength und xLength vertauscht waren in wasteBox, und das widerum lag daran, dass createFromXML die width und length falschrum in den Konstruktor gepackt hat.

Wieder etwas, was mit named-arguments anstelle von unzähligen unnamed args in constructors nicht passiert wäre.

closed 2022-08-21 by memoryleak47