hz-b/rayx #74: Consider removing ElementParams, SurfaceParams, etc. in favor of a simple struct for each Object within the shader.

memoryleak47, 2022-08-02:


memoryleak47, 2022-08-16:
There are 3 ideas on how to implement this:
1. "shallow": we do not change anything about the VulkanTracer stuff etc.
But we add structs for each Element-type both in the shader and in the CPP files.
We then write a function in the shader to reinterpret the ElementParams, SurfaceParams etc. to reconstruct the structs from it.
This way we can use conventient structs in the shader without much rebuilding, but we earn technical debt to be cleared up later.

2. "2 buffers": we add one double-buffer containing all data of all elements in sequence (the first double of each element would be its type).
And another buffer to express where which element starts.
This is similar to materialBuffer and materialIndexBuffer.
The advantage is that we only need 2 buffers to do everything, which is simple in VulkanTracer and in the shader.
But we need to compile things down to a double-stream to afterwards reinterpret everything back to structs in the shader.
This can probably be done using some kind of reinterpret_cast-version, or with unions in GLSL. (I'm confident I can code this reinterpreting in 1 hour).
Disadvantage: We need new buffers if compared to "shallow".

3. "n-buffers": we add a buffer for each element-type.
The advantage is that we don't need any reconstructions, as we can simply put an RZP-struct into the shader, which can directly be interpreted as this same RZP-struct without any effort.
Disadvantage: indices don't simply work for elements anymore, because element 0 could be the first RZP or the first PlaneMirror etc.
Disadvantage: we need to add many buffers, and even more buffers when we add new elements.

I think we should either
- implement "2 buffers"
- do nothing, if you think we have more pressing issues

memoryleak47, 2022-08-16:
A potentially even simpler idea would be to do "2 buffers" but merge them into one buffer.
First double would be the length of the ElementIndexBuffer,
then comes the content of the ElementIndexBuffer.
and then comes the content of the ElementBuffer (i.e. all elements reinterpreted to doubles concatenated together)

Atraxus, 2023-02-24:
@memoryleak47 will implement the 2. option from his comment. This could also make a rework of the objects possible, where each object comprises a geometry, a surface, and a struct/class for the element parameters. This would allow for better code reusability, modularity, and clarity.

memoryleak47, 2023-03-30:
done now using the Behaviour, Cutout and Surface systems!

closed 2023-03-30 by memoryleak47