hz-b/rayx #82: Implement Misalignment and then re-add ellipsoidmis tests.

memoryleak47, 2022-08-21:
Those are two tests with RML input file and a CSV-RayUI file.

[ellipsoid_ip_200ellipsmis.csv](https://github.com/hz-b/RAY-X/files/9389916/ellipsoid_ip_200ellipsmis.csv)
[ellipsoid_ip_200ellipsmis.rml.txt](https://github.com/hz-b/RAY-X/files/9389917/ellipsoid_ip_200ellipsmis.rml.txt)
[ellipsoid_ip_200mirrormis.csv](https://github.com/hz-b/RAY-X/files/9389918/ellipsoid_ip_200mirrormis.csv)
[ellipsoid_ip_200mirrormis.rml.txt](https://github.com/hz-b/RAY-X/files/9389919/ellipsoid_ip_200mirrormis.rml.txt)

This is not high priority right now.

memoryleak47, 2022-08-22:
I just added the test ellipsoid_ip_200mirrormis.
What's still missing is ellipsoid_ip_200ellipsmis.

Note that those tests .rml files only differ in the param misalignmentCoordinateSystem, which we don't support yet.

Atraxus, 2023-09-21:
Done

closed 2023-09-21 by Atraxus