hz-b/rayx #91: Building in release fails building googletest

hzbian, 2022-09-06:
```
cmake --build ./build --config Release --target all -j 10 --
[45/62] Building CXX object Extern/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
FAILED: Extern/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o 
/usr/bin/c++  -I/home/hzbian/RAY-X/Extern/googletest/googletest/include -I/home/hzbian/RAY-X/Extern/googletest/googletest -Wall -Wextra -Wpedantic -O3 -DNDEBUG -Wall -Wshadow -Werror -Wno-error=dangling-else -DGTEST_HAS_PTHREAD=1 -fexceptions -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -std=c++20 -MD -MT Extern/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o -MF Extern/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o.d -o Extern/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o -c /home/hzbian/RAY-X/Extern/googletest/googletest/src/gtest-all.cc
In Datei, eingebunden von /usr/include/c++/12/ios:40,
                 von /usr/include/c++/12/ostream:38,
                 von /usr/include/c++/12/bits/unique_ptr.h:41,
                 von /usr/include/c++/12/memory:76,
                 von /home/hzbian/RAY-X/Extern/googletest/googletest/include/gtest/gtest.h:57,
                 von /home/hzbian/RAY-X/Extern/googletest/googletest/src/gtest-all.cc:38:
In statischer Elementfunktion »static constexpr std::char_traits<char>::char_type* std::char_traits<char>::copy(char_type*, const char_type*, std::size_t)«,
    eingefügt von »static constexpr void std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_S_copy(_CharT*, const _CharT*, size_type) [mit _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]« bei /usr/include/c++/12/bits/basic_string.h:423:21,
    eingefügt von »constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_M_replace(size_type, size_type, const _CharT*, size_type) [mit _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]« bei /usr/include/c++/12/bits/basic_string.tcc:532:22,
    eingefügt von »constexpr std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::replace(size_type, size_type, const _CharT*, size_type) [mit _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]« bei /usr/include/c++/12/bits/basic_string.h:2171:19,
    eingefügt von »constexpr std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::insert(size_type, const _CharT*) [mit _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]« bei /usr/include/c++/12/bits/basic_string.h:1928:22,
    eingefügt von »constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator> std::operator+(const _CharT*, __cxx11::basic_string<_CharT, _Traits, _Allocator>&&) [mit _CharT = char; _Traits = char_traits<char>; _Alloc = allocator<char>]« bei /usr/include/c++/12/bits/basic_string.h:3541:36,
    eingefügt von »static std::string testing::internal::StreamingListener::UrlEncode(const char*)« bei /home/hzbian/RAY-X/Extern/googletest/googletest/src/gtest.cc:4882:27:
/usr/include/c++/12/bits/char_traits.h:431:56: Fehler: »void* __builtin_memcpy(void*, const void*, long unsigned int)« Zugriff auf 9223372036854775810 oder mehr Bytes an Offsets [2, 9223372036854775807] und 1 überdeckt möglicherweise 9223372036854775813 Bytes an Offset -3 [-Werror=restrict]
  431 |         return static_cast<char_type*>(__builtin_memcpy(__s1, __s2, __n));
      |                                        ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
cc1plus: Alle Warnungen werden als Fehler behandelt
[49/62] Building CXX object Tests/CMakeFiles/RAYX_tst.dir/cmake_pch.hxx.gch
ninja: build stopped: subcommand failed.
```

memoryleak47, 2022-09-20:
The problem is that googletest throws a warning, and your compiler has warnings-as-errors activated.
Is your compiler somehow configured to do this per default?
The build system of RAY-X should only generate -Werror if CI=YES is set as an environment variable, which I presume should not be the case for you.

hzbian, 2022-09-26:
This works now.

closed 2022-09-26 by hzbian

memoryleak47, 2022-10-02:
Nice! I added gtest as a submodule, that might have fixed it.