hz-b/rayx #92: Errors in Wiki

hzbian, 2022-09-06:
Add this to the required packages for Fedora:
`ninja-build gcc-c++ vulkan-loader-devel glslc`

And the order of the setup and compiliation commands building without IDE is wrong way round.

memoryleak47, 2022-09-21:
Thanks! I added the fedora packages.
Additionally I added a `compile.sh` for a better non-IDE experience.
It's `compile.sh --release` for the release mode accordingly.

closed 2022-09-21 by memoryleak47

hzbian, 2022-09-26:
Shouldn't it be the other way round?  A flag for debug and release otherwise?

memoryleak47, 2022-09-26:
> Nice! But shouldn't it be the other way round? A flag for debug and release otherwise?

I did it like Rusts cargo. But I don't really mind.
If you want to, I can change --release to be the default.