hz-b/rayx #93: h5 no such file or directory on -p

hzbian, 2022-09-07:
The ` -p`  option turns to the following error in Linux:

> Traceback (most recent call last):
>   File "/home/hzbian/RAY-X/build/bin/python/py_plot.py", line 160, in <module>
>     plotOutput(str(sys.argv[1]), int(sys.argv[2]))
>   File "/home/hzbian/RAY-X/build/bin/python/py_plot.py", line 146, in plotOutput
>     df = importOutput(h5_file)
>   File "/home/hzbian/RAY-X/build/bin/python/py_plot.py", line 23, in importOutput
>     with h5py.File(filename, 'r') as h5f:
>   File "/home/hzbian/RAY-X/build/bin/rayxvenv/lib64/python3.10/site-packages/h5py/_hl/files.py", line 533, in __init__
>     fid = make_fid(name, mode, userblock_size, fapl, fcpl, swmr=swmr)
>   File "/home/hzbian/RAY-X/build/bin/rayxvenv/lib64/python3.10/site-packages/h5py/_hl/files.py", line 226, in make_fid
>     fid = h5f.open(name, flags, fapl=fapl)
>   File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
>   File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
>   File "h5py/h5f.pyx", line 106, in h5py.h5f.open
> FileNotFoundError: [Errno 2] Unable to open file (unable to open file: name = 'METRIX_U41_G1_H1_318eV_PS_MLearn.h5', errno = 2, error message = 'No such file or directory', flags = 0, o_flags = 0)
> Traceback (most recent call last):
>   File "/home/hzbian/RAY-X/build/bin/python/py_plot_entry.py", line 24, in startPlot
>     output = subprocess.check_call(
>   File "/usr/lib64/python3.10/subprocess.py", line 369, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['/home/hzbian/RAY-X/build/bin/rayxvenv/bin/python3', '/home/hzbian/RAY-X/build/bin/python/py_plot.py', 'METRIX_U41_G1_H1_318eV_PS_MLearn.rml', '1']' returned non-zero exit status 1.
> [        TerminalApp.cpp:129] Error while running the python module: [ERR011]startPlot
> [        TerminalApp.cpp:129] Terminating...

memoryleak47, 2022-09-20:
Hey,

on which commit are you? please try the newest version.
what command did you execute?

hzbian, 2022-09-26:
Same problem on 368fde220a040ce7e7827e3c56e7c1c578336494.

memoryleak47, 2022-09-26:
Ah, it's an absolute path problem.
I'll look into it.

Until it's fixed you can copy BL_file.rml to your TerminalApp and do
`./TerminalApp -x -i BL_file.rml -p` instead.

memoryleak47, 2022-10-02:
Should be fixed now, please try again @hzbian

hzbian, 2022-10-06:
This works now. Shouldn't this issue be covered by a unit test?

memoryleak47, 2022-10-10:
Also, this error might just resolve itself once we plot within c++,
which is currently being implemented by Oussama.

So I'd say we'll wait a few weeks, and then re-check whether the issue remains.

OSMaxwell, 2022-10-21:
Hello there, 
An almost similar plot version is now live on dev branch all written in C++ . (Though it still calls py interpreter internally.)

Make sure that you have python3, python3-dev and python3-matplotlib (also through pip) installed for this.

hzbian, 2022-11-07:
This works now, no issue left. Thank you @OSMaxwell @memoryleak47 for the great work!

commit f2d8e1dd (2022-10-02): revamped path-canonicalization system (fixes #93)
closed 2022-11-07 by hzbian