hz-b/rayx #94: Multiplot Bug

Atraxus, 2022-10-04:


OSMaxwell, 2022-10-04:
[image]

OSMaxwell, 2022-10-18:
[rml_file_and_dat.zip](https://github.com/hz-b/RAY-X/files/9812163/rml_file_and_dat.zip)

memoryleak47, 2022-10-23:
There are two issues here:
Petes issue: The objects 1 to 5 should be considered as one RZP for Multi-plotting.

The problem with that is that they are just separate objects in the RML file (not even grouped).
So the Plotter does not know, which of them belong together.

Jannis' issue: How can an extraParam have the same digit twice?

I dug into the shader and found out that the shader actually has some kind of self-intersection.
The situation is as follows:
a ray hit's object A: traceRZP is called.
Afterwards, this ray hits object B: traceRZP is called, but we get an early return because the wasteBox fails.
-> from the point-of-view of main.comp we had another collision, namely B (because closestElementIndex was changed),
but the extraParam was not changed because no actual collision happened due to wasteBox failure.

So at this point the Ray actually collides with A again, which is not detected as self-intersection anymore, as closestElementIndex was changed.
-> the extra param gets the same digit (A) twice.

Atraxus, 2022-10-24:
Concerning the first part: As far as I can see, the RZPs are grouped in the test_multi.rml (`<group>...</group>`)

The second part: Top, good to hear that's fixed now. This only influenced the extraParam, right?

memoryleak47, 2022-10-25:
Oh my bad, they really are grouped.
Multiplotting just doesn't handle grouping, as grouping currently is only used to calculate positions & orientations of objs,
and then the group information is not passed on.

Second part:
I think the bug doesn't only affect extraParam.
The shader sometimes allowed self-intersections, even though it's intended to disallow them.
So after this fix, each ray which hit the same object twice directly after each other,
will have to take a different route, and hence will look quite different.

Btw, do we have any plans on supporting self-intersections?

OSMaxwell, 2022-10-28:
new multiplot figure:
[image]

closed 2022-10-28 by Atraxus

pforson, 2022-10-28:
61 -> 65 look good. I'll plot them together and compare with measured data. I'll check if 6 makes sense.

Atraxus, 2022-10-29:
Closed this issue as the 'repeating numbers' bug is now fixed. See #111 for strange RZP behaviour