hz-b/rayx #96: Wrong version number

hzbian, 2022-10-06:
I am on 3d951150684573fc7c76e1c91ee54f0b14b09b21, however:

```
[         CommandParser.h:35] 
          
        ╔═══╗╔═══╗╔╗  ╔╗     ╔═╗╔═╗
        ║╔═╗║║╔═╗║║╚╗╔╝║     ╚╗╚╝╔╝
        ║╚═╝║║║ ║║╚╗╚╝╔╝      ╚╗╔╝ 
        ║╔╗╔╝║╚═╝║ ╚╗╔╝ ╔═══╗ ╔╝╚╗ 
        ║║║╚╗║╔═╗║  ║║  ╚═══╝╔╝╔╗╚╗
        ╚╝╚═╝╚╝ ╚╝  ╚╝       ╚═╝╚═╝ HZB 2022.
        
[         CommandParser.h:44]    RAY-X Terminal Application 0.4.0.0
         GIT: 368fde2
         BUILD: 2022-09-26T18:54:09
         ```
Also I've compiled it today …

OSMaxwell, 2022-10-21:
Hi there,
We have tested this and cannot reproduce the error.
Make sure that you are running the correct exec. 
The build is now under ../build/debug/bin or ../build/release/bin

hzbian, 2022-10-21:
Sorry, I cannot build anymore, because of #102.

OSMaxwell, 2022-10-28:
We are working on fixing the dev-branch. Seems not to be working for some...

Atraxus, 2022-10-29:
Dev branch should be working again, could you try this again, to see if the issue still persists?

hzbian, 2022-11-07:
Yes, this works now, issue is gone, thanks!

closed 2022-11-07 by hzbian