hz-b/rayx #102: vk_mem_alloc.h No such file or directory

hzbian, 2022-10-21:
When I compile now (as of 05a5d6a0ffd13e35010060d3937bdefc3268bb53), i get the following error:

> RAY-X/Intern/RayCore/src/VulkanEngine/VulkanEngine.h:3:10: schwerwiegender Fehler: vk_mem_alloc.h: Datei oder Verzeichnis 
> nicht gefunden
>     3 | #include <vk_mem_alloc.h>
>       |          ^~~~~~~~~~~~~~~~
> Kompilierung beendet.
> [37/69] Building CXX object Intern/RayCore/CMakeFiles/RayCore.dir/src/Tracer/CpuTracer.cpp.o
> ninja: build stopped: subcommand failed.
>

It happens both on release and debug.

OSMaxwell, 2022-10-21:
Yes, I should write that soon in the docs. You need at least Vulkan 1.3 for the new addition as of 5fb016c47f0919244bb5487022f6bcd30c777ce9. 
Find it here : 
https://vulkan.lunarg.com/sdk/home#linux

Remove your old vulkan-sdk first before upgrading.
Overwriting vulkan-sdk can break stuff..

Atraxus, 2022-10-21:
I have the same issue and I use Vulkan version 1.3.224.1.
So, it seems to be something different than the version...

OSMaxwell, 2022-10-22:
Can it be Windows related? on the CI "Docker" machine it seems to be found...?

hzbian, 2022-10-28:
Nope does also not work on the docker version. Furthermore, I use Vulkan 1.3 on my desktop now, I still have the same issue.

Atraxus, 2022-10-28:
We talked about the issue in the meeting today. It seems not all Vulkan installations bring the vma headers, even if using v1.3+
@memoryleak47 reported, that there is an extra package on arch that he installed to make it work

hzbian, 2022-10-28:
Is there the possibility to directly include this directly in our build repo?

https://github.com/GPUOpen-LibrariesAndSDKs/VulkanMemoryAllocator

OSMaxwell, 2022-10-28:
We did that on a different branch but now that the sdk offers that so we want to get away from that. Please try now :)
____
- If you are linking vulkan-sdk through shell with VULKAN-SDK, This should be working out of the box.

- If you are on **Ubuntu** or similar debian-based, please install **vma** (Something similar is also on arch)

- If you are on **WIndows**,  on the vulkan-sdk installation, check the **"Vulkan Memory Allocator header"** option

Atraxus, 2022-10-29:
This should be fixed by 60007e4, where vma is included as a submodule

closed 2022-10-29 by Atraxus

hzbian, 2022-11-07:
This is fixed now. Is it merged to development yet?

Atraxus, 2022-11-07:
Yes, this is on the dev branch.