hz-b/rayx #103: NumPy missing

hzbian, 2022-10-21:
Compiling in docker, I get the following error now (as of 05a5d6a0ffd13e35010060d3937bdefc3268bb53):
```CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found
  version "3.10.7")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindPython/Support.cmake:3166 (find_package_handle_standard_args)
  /usr/share/cmake/Modules/FindPython3.cmake:490 (include)
  Intern/RayCore/CMakeLists.txt:192 (find_package)```

Atraxus, 2022-10-21:
I got the same error after reinstalling Python, and after installing NumPy I don't get the error anymore. 
Maybe you don't have it installed in the docker?

OSMaxwell, 2022-10-28:
Hola! 
You need python3 python3-matplotlib python3-numpy (pip or apt) 
numpy usually comes bundles with matplotlib but just to be safe.

hzbian, 2022-10-28:
Okay, this fixed the issue.

closed 2022-10-28 by hzbian