hz-b/rayx #111: RZP bug

Atraxus, 2022-10-28:


Atraxus, 2022-10-28:
[image]
Issues marked blue

treegardel, 2022-11-11:
Blue framed plots: Here z should be plotted over x because it is a reflective element.
This applies to all reflective optics:
- All mirrors and gratings
For transmissive optics, y should be plotted over x, i.e. Image Plane, Slit, etc.

Atraxus, 2022-11-11:
Blue arrows might be fixed by recent commit

Atraxus, 2022-11-14:
Light rays being reflected from one RZP to another doesn't seem to be an issue anymore.

Atraxus, 2022-11-25:
Blue arrow plots reappeared with recent commits. I'm going to check this out

Atraxus, 2023-02-23:
Fixed by the new plotting system

closed 2023-02-23 by Atraxus