hz-b/rayx #113: Single-Plot only generates one point

memoryleak47, 2022-10-28:
The file given here:
https://github.com/hz-b/RAY-X/issues/94
only generates a single point when ran with -p

memoryleak47, 2022-10-28:
[image]
that's how it looks for me.

Atraxus, 2022-10-31:
I get only one point, as far as I can see (this is on a6d3e3d4, the single-plot branch doesn't output anything)

[image]

OSMaxwell, 2022-11-09:
This might be a bug in the tracing : 
As we're emulating RAY-UI : We need to look for rays that crossed the longest length 1234 etc
Here's an output of test_multi.rml
Left : extraParam, Right: amount
```
[            Plotter.cpp:157] 0:5840
[            Plotter.cpp:157] 1:4594
[            Plotter.cpp:157] 2:53862
[            Plotter.cpp:157] 3:62308
[            Plotter.cpp:157] 4:53617
[            Plotter.cpp:157] 5:4420
[            Plotter.cpp:157] 6:1107039
[            Plotter.cpp:157] 54:1
[            Plotter.cpp:157] 61:49366
[            Plotter.cpp:157] 62:187286
[            Plotter.cpp:157] 63:234534
[            Plotter.cpp:157] 64:187468
[            Plotter.cpp:157] 65:49660
[            Plotter.cpp:157] 212:1
[            Plotter.cpp:157] 454:1
[            Plotter.cpp:157] 612:2
[            Plotter.cpp:157] 654:1
```

You can see that the ones that meet 6 (ImagePlane) with the longest path are not that much!!

OSMaxwell, 2022-11-09:
Here's a trace when looking only for the Imageplane being last as in 6XXXXX is accepeted
[image]

However this doesn't look like RAY-UI
[image]

The fix also breaks David's beamline which used to be correct when emulating the "sequential tracing" in plots

memoryleak47, 2022-11-10:
>  As we're emulating RAY-UI : We need to look for rays that crossed the longest length 1234 etc

Can you explain what you mean with that?
I'm under the impression "single-plot" should render all rays ending in the ImagePlane, not only those taking the longest path to it, right?

OSMaxwell, 2022-11-11:
Fixed in f48389a921b4ce6a2b7b92e3b71537e5307893aa . Use LONGEST_PATH if you want to revert back to old footprints

closed 2022-11-11 by OSMaxwell