hz-b/rayx #116: missing matplotlib

treegardel, 2022-11-03:
When using the commandline option -p I got the following error:

`[             Plotter.cpp:18] Plotting...
ModuleNotFoundError: No module named 'matplotlib'
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error loading module matplotlib!
Aborted (core dumped)`

There must be something wrong with my Python Matplotlib, right?

Atraxus, 2022-11-03:
Are you sure you installed the matplotlib Module?
(maybe try `python3 -m pip install matplotlib` to be sure)

treegardel, 2022-11-03:
Thanks @Atraxus; with your command for installing the lib it works now.

treegardel, 2022-11-04:
Strange - now it fails again with another error:
`Starting RAY-X (b74497dd, release)
Tracing File: "../../../../SimpleTest/MultiRun/METRIX_U41_G1_H1_318eV_PS_MLearn_AutoDIA_org1.rml"
[             Plotter.cpp:18] Plotting...
terminate called after throwing an instance of 'std::runtime_error'
  what():  Call to title() failed.
Aborted (core dumped)`
(matplotlib is installed via pip)

OSMaxwell, 2022-11-09:
Plot results might be still broken. But it should work now..?

treegardel, 2022-11-11:
Command line option "-p -m" still not working...

`Starting RAY-X (0d5187d2, release)
Tracing File: "../../../../SimpleTest/MultiRun/METRIX_U41_G1_H1_318eV_PS_MLearn_AutoDIA_org1.rml"

[             Plotter.cpp:31] Plotting...
terminate called after throwing an instance of 'std::runtime_error'
  what():  Call to subplot() failed.
Aborted (core dumped)`

OSMaxwell, 2022-11-11:
matplotlib-cpp works so far with matplotlib==3.1.2
Installing through pip gives a more recent version.

Solved through matplotlib downgrade

closed 2022-11-11 by OSMaxwell

Atraxus, 2022-11-14:
I pushed a fix for this issue. For me, multi-plotting works now even with Python v3.10.