hz-b/rayx #118: order not clear, when running a directory containing rml files

treegardel, 2022-11-04:
As a user, I would expect the RML files to be processed in alphabetical order. For me it looks like a random order, even after I "touched" them all one after the other:

Here is the screenshot of my directory:
[image]

A nice feature would be to display the file that is being traced in the terminal. It is displayed in -V mode, but then so many more things are displayed.

memoryleak47, 2022-11-04:
fixed in 209b7e7.

treegardel, 2022-11-04:
I just tested it - I see the filename, but the order is still random. Didn't we want to change this so that the files are at least processed alphabetically?

Atraxus, 2023-03-15:
RAY-X traces files in a directory in alphabetical order for me. @treegardel, could you check again if this behavior persists?

closed 2023-08-07 by Atraxus