hz-b/rayx #120: Rethink optical element data structure

Atraxus, 2022-11-07:


Atraxus, 2022-11-07:
First two questions for the graphic:
- Is my understanding of how we calculate the precise intersection correct?
- What parameters are used in the different steps? Can there be a clear division?
- Does this division make sense? Would it really improve the implementation?

treegardel, 2022-11-07:
I don't know how the intersection calculation is implemented at the moment.

In RAY-UI the intersection calculation works a bit more direct: Here the intersection is calculated immediately with the curved surface, then it is checked if the intersection point is inside the boundary - if it is outside, the ray is thrown away.

However, I think that your method of first looking to see if the "waste box" (which should actually be called a hit box) is hit could be more efficient.

However, we would still have to introduce a small improvement: Instead of a flat waste box, we need a 3-dimensional box around the curved surface. If this box is hit, the intersection calculation continues inside the box.

closed 2023-02-23 by Atraxus