hz-b/rayx #122: Trapezoidal geometrical shape: does work only in one direction

treegardel, 2022-11-17:
Simple set up with a plane mirror:
[image]
In case parameter "totalWidth" is set to 50 and "
[MatrixPMtrapez50-10.rml.txt](https://github.com/hz-b/RAY-X/files/10034823/MatrixPMtrapez50-10.rml.txt)
totalWidthB to 10 it looks reasonable:
RML-File: [MatrixPMtrapez50-10.rml.txt](https://github.com/hz-b/RAY-X/files/10034777/MatrixPMtrapez50-10.rml.txt)
[image]
But in case the trapezoidal shape is rotated by 180°: "totalWidth" = 10; "totalWidthB" = 50 it looks wrong:
RML-File: [MatrixPMtrapez50-10.rml.txt](https://github.com/hz-b/RAY-X/files/10034825/MatrixPMtrapez50-10.rml.txt)
[image]
There is no trapezoidal shape visible. It is a simple rectangular shape with a width of 10.
This is exactly the case we need at the RZP Array: totalWidthB (width upstream) > totalWidth (width downstream)

Atraxus, 2022-11-18:
[image]

This seems to be how it is implemented in the shader. Maybe the formula use has a bug in it. I will check that

Atraxus, 2022-11-24:
[image]
Correcting my previous comment: This is how it is implemented (ignore points 1 and 2)

After resolving some issues in the shader, I get the following outputs
[image]
For totalWidthA (LengthA) = 10 and totalWidthB (LengthB) = 50

[image]
For totalWidthA (LengthA) = 50 and totalWidthB (LengthB) = 10

This resolves the issue