hz-b/rayx #127: Current development build can't trace METRIX beamline:

treegardel, 2022-12-02:
Here is the output:
`baumgaertel@notexray04:~/RAY/RAY-X/build/bin/release$` ./TerminalApp -i ../../../../SimpleTest/MultiRun/METRIX_U41_G1_H1_318eV_PS_MLearn_AutoDIA_org4.rml -p
Starting RAY-X (2de64d30, release)
Tracing File: "../../../../SimpleTest/MultiRun/METRIX_U41_G1_H1_318eV_PS_MLearn_AutoDIA_org4.rml"
[    RunCommandBuffer.cpp:36] Fatal : VkResult fail!
[    RunCommandBuffer.cpp:36] Error code: -4, look up at https://www.khronos.org/registry/vulkan/specs/1.3-extensions/man/html/VkResult.html
[    RunCommandBuffer.cpp:36] Terminating...'

memoryleak47, 2022-12-02:
It turned out that this error does not come up with fewer rays.
We thought that we had a graceful error message if given too many rays.
But apparently that doesn't work as of now.

@treegardel can you provide the .rml file?

treegardel, 2023-01-06:
Before closing the issue: here the current METRIX - file, which  is used by the ML-Group
[METRIX_U41_G1_H1_318eV_PS_MLearn_MG.rml.txt](https://github.com/hz-b/RAY-X/files/10358811/METRIX_U41_G1_H1_318eV_PS_MLearn_MG.rml.txt)

closed 2023-01-06 by treegardel