hz-b/rayx #128: METRIX Beamline: multiplot gives 3 additional plots on last element which are not explainable

treegardel, 2023-01-06:
Running the METRIX Beamline: 
[METRIX_U41_G1_H1_318eV_PS_MLearn_MG.zip](https://github.com/hz-b/RAY-X/files/10359010/METRIX_U41_G1_H1_318eV_PS_MLearn_MG.zip)
and looking at the multiplot result:
[image]
Shows 4 Plots, **but only the last plot "987654321" should appear**. The others are not explainable where they should come from.

[image]

RAY-UI clearly shows that only at the first element (ASBL, slit) rays are "lost" due to absorption. After that, all rays pass sequentially over the other elements. So that only plot 987654321 is correct, these are the only rays that reach the last element (ImagePlane).

Plot "98765432": It looks like the rays absorbed at the slit are traced further with the status of being a new source.

OSMaxwell, 2023-01-18:
Is the split too small compared to the Cylinder maybe? 2 is Cylinder which means 98765432 start from cylinder and ignore slit

treegardel, 2023-01-18:
For me, 98765432 are exactly the rays that do not go through the aperture but are absorbed by the apertures, hence the white area in the middle. So they should not be traced further. The rays that pass through are correctly represented in 987654321.
Is it not possible to debug this relatively easily in -x mode?
It would also be interesting to see whether we still get 98765432 if we set the aperture so small that no rays come through at all. I'll try it out tomorrow.

Translated with www.DeepL.com/Translator (free version)

treegardel, 2023-01-20:
Here is the result: I closed the front end aperture totally (by setting the slit size to zero/zero). All rays should be blocked, as in RAY-UI, but we still get the following plots:
[image]
There must be a bug.

Atraxus, 2023-03-04:
We changed our plotting

closed 2023-03-04 by Atraxus