hz-b/rayx #131: vkCreateDevice not working with Intel Iris Xe Graphics

Teddish, 2023-01-17:
Unfortunately the Iris Xe Graphics card doesn't support shaderFloat64 and shaderInt64 and thus vkCreateDevice will throw a VK_ERROR_FEATURE_NOT_PRESENT in [PickDevice.cpp](https://github.com/hz-b/RAY-X/blob/84f2089935ed002706aa3d4a170494b02acac457/Intern/RayCore/src/VulkanEngine/Init/PickDevice.cpp#L168).
Setting the variables [shaderInt64](https://github.com/hz-b/RAY-X/blob/84f2089935ed002706aa3d4a170494b02acac457/Intern/RayCore/src/VulkanEngine/Init/PickDevice.cpp#L148) and [shaderFloat64 ](https://github.com/hz-b/RAY-X/blob/84f2089935ed002706aa3d4a170494b02acac457/Intern/RayCore/src/VulkanEngine/Init/PickDevice.cpp#L147) to false will allow  Device Creation without any error but my understanding is that the shader then needs to be adapted. If possible it would be nice for compatibility reasons.

Additional info can be found [on the Intel homepage](https://www.intel.com/content/www/us/en/support/articles/000089817/graphics.html) and on the Khronos Homepage:
[shaderInt64](https://registry.khronos.org/vulkan/specs/1.3-khr-extensions/html/chap38.html#features-shaderInt64)
[shaderFloat64](https://registry.khronos.org/vulkan/specs/1.3-khr-extensions/html/chap38.html#features-shaderFloat64)
[vkCreateDevice](https://registry.khronos.org/vulkan/specs/1.3-extensions/man/html/vkCreateDevice.html)

OSMaxwell, 2023-01-18:
Good find,
though I think this might end up being not supported as we aim for high precision that Int/float32 might not offer

@Atraxus ?..

Atraxus, 2023-01-18:
Oussama is right; single precision is currently not supported. We already discussed a single precision mode for higher performance, but it is low on our priority list. We will discuss this in the next meeting and see if it's worth looking at that feature earlier than we intended to.

Teddish, 2023-01-19:
Thanks a lot for the discussion and explanation. I'll try to stick to a PC with a GPU that supports these features for now. :)

Atraxus, 2023-01-20:
We just discussed this and concluded that we would like to try this as an experiment, but compared to the time investment, it might not yield good results. We will look at it, but current issues are more pressing. 

I am closing this issue for now.

closed 2023-01-20 by Atraxus