hz-b/rayx #132: RAY-X Docker Support

Atraxus, 2023-01-20:
@hzbian, can you tell me what you tried so far to get a docker running?

hzbian, 2023-01-20:
Tested all of them with enroot (https://github.com/NVIDIA/enroot)
nvidia-vulkan image https://hub.docker.com/r/nvidia/vulkan
nvidia-cuda and installing vulkan from Repo 
Fedora 37 installing vulkan from Repo
With all kinds of pass-through variables (cf. https://developer.nvidia.com/blog/gpu-containers-runtime/, NVIDIA_DRIVER_CAPABILITIES)

`
FROM fedora:37

RUN dnf update -y; dnf clean all
RUN dnf install -y cmake git gcc gdb vulkan vulkan-tools vulkan-validation-layers hdf5-devel ninja-build gcc-c++ vulkan-loader-devel glslc python3-devel; dnf clean all

RUN git clone --recurse-submodules https://github.com/hz-b/RAY-X.git -b broken_includes
RUN cd RAY-X && git checkout 60007e41405e79fedd2f5249b14dd43bfd9328ad && cd ..

RUN dnf install -y python3-pip
RUN pip install -U numpy

RUN cd RAY-X && sh compile.sh
RUN cd RAY-X/build/bin/debug
`

But always vulkaninfo already throws an error (driver incompatible) or GPU is not shown.

Atraxus, 2023-02-06:
The only solution I found so far is to install everything necessary for RAY-X with a Dockerfile and only install Vulkan after the docker is run. This fixes the "incompatible driver"-bug, and with `docker save` or similar functionality, the image could be moved to other machines. 

I'm currently looking into why the NVIDIA Vulkan image fails.

Atraxus, 2023-02-10:
After more trial and error, I can't find another way to get this running. I am closing this issue now, as there is a solution, even if it is not a good one. Please let me know if you need any help.

closed 2023-02-10 by Atraxus