hz-b/rayx #133: Slit needs a colliding surface around it.

memoryleak47, 2023-01-27:
A slit has a beamstop- width, height and total- width, height.
Rays hitting the beamstop are absorbed, and rays around it pass through.

In Ray-UI this worked out, because rays which missed the slit were discarded anyways - but for Ray-X this doesn't work.
If a ray slightly passes a slit, it will skip the slit and hit the next element.

options:
- add some kind of "wrapper"- width, height for the colliding surface around the slit
- automatically have an infinite wrapper size
- leave it as it is?

treegardel, 2023-02-09:
The METRIX Beamline is now running with the infinite wrapper size. But do we really want to have it like this? It's for really an optical element with a very big influence. We should introduce wrapper parameters. One is called totalWidth the other totalHeight. I will add them also in RAY-UI with some automatically created values somehow based on the actual slit size.

memoryleak47, 2023-02-10:
Right, last meeting we decided to go with the infinite wrapper size.
I don't yet see a reason why we want to change this to the "wrapper"-width approach now as Pete seemed quite happy with it.

I'm in favor of closing this, and re-opening a new issue incase someone wants a different Slit?

closed 2023-02-22 by Atraxus