hz-b/rayx #134: Use cases and missing things when using plot.py for visualisation of the traced rays

treegardel, 2023-02-09:
Use case:
- I only want to plot certain elements
- I want to see the name of the element -> Metadata in the h5 output (name, type and world coordinates, size of element ... ), or the whole rml-file?
- I want to create my own histogram by giving the limits and the size of the bits in x an d y (or x and z)
- I miss the unit in the description of the axis, following the DIN it is " / mm" (and also in legend: "# / bit")
- I would like to see a spot size analysis (Full Width Half Maximum, or as a minimal approach: RMS value in x and y)

Atraxus, 2023-02-10:
- [ ] Plots as multiple tabs in one window
- [ ] Support for -p option in RAY-X

Atraxus, 2023-03-04:
As discussed, @flxai and @pforson will take over and continue improving the plotting system

flxai, 2023-03-31:
Mostly resolved by https://github.com/hz-b/RAY-X/commit/2219a42d519432c2a0bfd95f0a4d8845abf3df50. Only thing left to do is to incorporate the Jupyter notebook (e.g. using the `-p` flag). As hinted in the notebook itself, it should be started with the environment variables `$RAYX_FILE` and/or `$RAYX_DIR`. Jupyter Lab can e.g. called like this:

```console
$ RAYX_FILE=/path/to/output.h5 RAYX_DIR=/path/to jupyter lab
```

This would select `/path/to/output.h5` in the lower dropdown menu, while still making all `.h5` files within `/path/to` available (using extended globbing `{,**/}*.h5`).

Manual intervention appears to be still necessary, since the user has to click on <kbd>Kernel</kbd> › <kbd>Restart Kernel and Run All Cells...</kbd> › <kbd>Restart</kbd> to run it, as detailed in the notebook as well.

Atraxus, 2023-06-01:
- [ ] -p option

Atraxus, 2023-08-07:
Is this still an active issue? I think we wanted to open the jupyter notebook with the -p option. Is this still relevant, or can we close this issue?

Atraxus, 2023-09-21:
-p is supposed to open the python plotting script not the notebook.

commit 2219a42d (2023-03-31): add jupyter notebook for visualization

resolves #134
closed 2023-09-21 by Atraxus