hz-b/rayx #171: Spherical RZP implementation

pforson, 2023-08-07:
In order to trace a spherical RZP RAY-X needs additional parameters from RAY-UI.
I have attached several rml files which I need to trace using RAY-X (they all work in RAY_UI).

From what I can tell, the following needs to be implemented in RAY-X (this list may not be complete and may be inaccurate):

- meridonal beam divergence (<param id="meridionalIncidenceBeamDivergence" enabled="T">45</param>)
- distance of divergences origin (<param id="meridionalIncidenceFocusDistance" enabled="T">90</param>)
- curvature type - spherical (<param id="curvatureType" comment="spherical" enabled="T">2</param>)
- Long Radius R (<param id="longRadius" enabled="T">4000</param>)
- Type of imaging (<param id="imageType" comment="point to horizontal divergent line" enabled="T">5</param>)
- Grating Line Profile (<param id="lineProfile" comment="unknown" enabled="T">3</param>)

The priority of this is high as I need the traces asap! :)

[spherical_RZP_rayx.zip](https://github.com/hz-b/rayx/files/12280179/spherical_RZP_rayx.zip)

Atraxus, 2023-08-07:
Updated the ticket to fit our templates.

treegardel, 2023-08-10:
Concerning Parameters
- meridional beam divergence (<param id="meridionalIncidenceBeamDivergence" enabled="T">45</param>)
- distance of divergences origin (<param id="meridionalIncidenceFocusDistance" enabled="T">90</param>)
are "Design Parameters", parameters which are used to derive a geometrical parameter. In this case the parameter  <param id="elementOffsetZ" enabled="F">0</param> is derived.

Unfortunately, RAY-UI and the current RML version do not strictly distinguish between Design and Geometrical relevant Parameters.

In this case I would proceed as follows: @pforson sets parameter 
<param id="elementOffsetZType" comment="by beam divergence and distance " enabled="T">1</param>
to
<param id="elementOffsetZType" comment="manual " enabled="T">0</param>
and
 <param id="elementOffsetZ" enabled="T">19.1521</param>
 (the value, which was derived)
 An in RAYX we need to implement this offset as a kind of misalignment.
 
 **Or:**
 @pforson move the Z-Translation simply into the misalignment (first translation the rotation)
 This would be se simplest way. Needs to be tested in RAY-UI as we still work on the RZP misalignment.

treegardel, 2023-08-10:
Concerning Parameters:
- curvature type - spherical (<param id="curvatureType" comment="spherical" enabled="T">2</param>)
- Long Radius R (<param id="longRadius" enabled="T">4000</param>) In RAY-UI, the RZP is the only element in which the curvature type can be changed within the element parameters. Otherwise it is done via the type of element. 

To become compatible with RAYX, I plan to change RAY-UI so that in the case of curvationType = "spherical" only the parameter "radius" is saved in the rml file. 
This should then be easy to read in RAYX, right?

treegardel, 2023-08-10:
Concerning Parameter
- Type of imaging (<param id="imageType" comment="point to horizontal divergent line" enabled="T">5</param>)
In this case, the 2-dimensional line density must be calculated slightly differently. Has this possibility already been prepared for implementation in RAYX?
If "yes", I see no problem in taking over the corresponding formula from RAY-UI (Fortran).

treegardel, 2023-08-10:
Last Parameter:
- Grating Line Profile (<param id="lineProfile" comment="unknown" enabled="T">3</param>)
We do not have to change anything in RAYX. 
@pforson: What was the value of this parameter in your other simulations?

memoryleak47, 2023-08-15:
> Type of Imaging

As of commit https://github.com/hz-b/rayx/commit/139bc9471281cca075943e203ffb6b6acace857e
"point to horizontal divergent line", "point to horizontal line" and "point to point" are supported.

Atraxus, 2023-09-21:
@EnricoAhlers and @pforson will check the current state on this

pforson, 2023-11-06:
Tested and is working as intended. Design Parameters will be added later in development.

closed 2023-11-06 by pforson