hz-b/rayx #180: RZP alignment issue

pforson, 2023-09-21:
**Describe the bug**
When tracing an RZP (planar or spherical) the traced result matches RAY-UI and experiment data in the case where no misalignment takes place. If misalignment is applied the resulting detector images are incorrect. This occurs not only when tracing multi-RZPs, but also regular single-RZPs.

**To Reproduce**
Steps to reproduce the behavior:
I will produce a few rmls which demonstrate this issue and add them here.

**Expected behavior**
I will produce plots which show the incorrect behaviour and the correct behaviour.

**Screenshots**
To follow...

**Desktop (please complete the following information):**
 - Ubuntu 22.04
 - CPU or Quadro T1000 GPU

**Additional context**
This fix is required for the Master Thesis of Enrico and therefore should have a high priority

memoryleak47, 2023-09-21:
I think the last state on this issue was, that we were waiting for you to test translational misalignment on the misalignment-rzp branch from a few weeks ago.

treegardel, 2023-09-21:
To be able to work on this further, we definitely need real-world results as well: x-translation scan of a plan single RZP and the camera image. With this, we first have to see whether RAY-UI calculates correctly. @pforson do you have such data from your measurements?

pforson, 2023-09-21:
Yes I have this experiment data. I'll try and compile everything as best I can to demonstrate the issue and update.

Atraxus, 2023-09-27:
@pforson and @treegardel will meet and discuss this further

pforson, 2023-09-28:
updated the original issue with screenshots - My opinion is that the change we made to RAY-X has not worked and it should be reverted back to the current RAY-UI method.

However - RAY-UI is also not correct as the Mn part only ever changes shape perfectly horizontally, which is incorrect.

treegardel, 2023-09-28:
@pforson please can you also add the rml-files for both cases, or a link if they are in another issue already? And do you also have a real case picture with no translation error?

pforson, 2023-09-28:
@treegardel I can't upload the rmls here (unsupported file format). I will upload in Mattermost.
Other examples, including a case with no translation error are in mattermost (dev channel).

treegardel, 2023-09-28:
[test_singleRZP.zip](https://github.com/hz-b/rayx/files/12752296/test_singleRZP.zip)
@pforson is it possible to transform the pixel units into mm in the real picture? (only for a "rough" estimation if this fits somehow)

pforson, 2023-10-04:
[image]

RAY-X development branch - looks like RAY-UI. This is closer to the the reality than the changes that were implemented in the misalignement-rzp branch. Therefore this branch should not be merged into development.

Further modifications to attempt to make the trace look more like the experiment images will be discussed.

pforson, 2023-11-06:
Tested and with an offset in the Z-axis traces equivalent to Ray-UI. This feature is currently in use and if any issues arise I will re-open.

closed 2023-11-06 by pforson