hz-b/rayx #181: SYCL Experimentation

Atraxus, 2023-09-21:
Testing if SYCL is a viable alternative for Vulkan computing.

memoryleak47, 2023-09-21:
Pro:
- We can use standard C++ in the "shader"
-> We can merge the shader code with the non-shader and Shared code
- We have an API that abstracts over GPU and CPU
-> We don't need a separated CpuTracer & GpuTracer

Contra:
- SYCL is probably not as broadly supported as Vulkan (yet?)
-- We hope that rusticl helps reaches its goal of general OpenCL support (for Linux) 
- Is SYCL as fast as Vulkan? After all it is much more high-level

memoryleak47, 2023-10-10:
After looking into SYCL for a bit, we've decided not to port our code to SYCL (yet).
Reasons:
1. SYCL doesn't seem to be stable everywhere yet (in particular, it doesn't run on my machine as of now)
2. SYCL doesn't support "full C++"
We can't use function pointers in SYCL, which prevents us from using GLM in the shader code currently.
Also things like dynamic dispatch and generally heap allocations (which is fair for a GPU language).
Still, these restrictions prevent us from implementing our code in standard C++ and then "simply" running this code on the GPU.

closed 2023-10-10 by memoryleak47