hz-b/rayx #187: Google Style Guide

Atraxus, 2023-10-04:
@OSMaxwell will do some research and check if we want to adopt additional rules from the Google style guide for our project. Further, automatically checking source and header files for style guide adherence might be interesting.

OSMaxwell, 2023-10-27:
Couple of useful notes that I had about this:
_______
- No forward declaration

- Define functions inline only when they are small, say, 10 lines or fewer.

- 
```
// Forbidden -- This pollutes the namespace.
using namespace foo;
```

- Constructors should never call virtual functions!

- Do not define implicit conversions.
```
class Foo {
  explicit Foo(int x, double y);
  ...
};
```

```
void Func(Foo f);

Func({42, 3.14});  // Error
```

- Use a struct only for passive objects that carry data; everything else is a class.
- Prefer to use a struct instead of a pair or a tuple whenever the elements can have meaningful names.

- Within each section, prefer grouping similar kinds of declarations together, and prefer the following order:

    Types and type aliases (typedef, using, enum, nested structs and classes, and friend types)
    (Optionally, for structs only) non-static data members
    Static constants
    Factory functions
    Constructors and assignment operators
    Destructor
    All other functions (static and non-static member functions, and friend functions)
    All other data members (static and non-static)
    
    
    
- Prefer to return by value or, failing that, return by reference. Avoid returning a pointer unless it can be null.

Atraxus, 2023-10-31:
After long discussions, we close this for now