hz-b/rayx #193: rayx-ui fails due to buffer with zero size

memoryleak47, 2023-10-11:
When running the rayx-core/tests/input/toroid.rml on commit 0d53c8a5, and then rendering the resulting .h5 file with rayx-ui, it fails on startup.

The .rml and .h5 files are stored in the attached [zip file](https://github.com/hz-b/rayx/files/12869320/zippy.zip).

This is my full output:
```
available extensions:
	VK_KHR_device_group_creation
	VK_KHR_display
	VK_KHR_external_fence_capabilities
	VK_KHR_external_memory_capabilities
	VK_KHR_external_semaphore_capabilities
	VK_KHR_get_display_properties2
	VK_KHR_get_physical_device_properties2
	VK_KHR_get_surface_capabilities2
	VK_KHR_surface
	VK_KHR_surface_protected_capabilities
	VK_KHR_wayland_surface
	VK_KHR_xcb_surface
	VK_KHR_xlib_surface
	VK_EXT_acquire_drm_display
	VK_EXT_acquire_xlib_display
	VK_EXT_debug_report
	VK_EXT_debug_utils
	VK_EXT_direct_mode_display
	VK_EXT_display_surface_counter
	VK_EXT_surface_maintenance1
	VK_EXT_swapchain_colorspace
	VK_KHR_portability_enumeration
	VK_LUNARG_direct_driver_loading
required extensions:
	VK_KHR_surface
	VK_KHR_xcb_surface
	VK_EXT_debug_utils
Device count: 1
physical device: AMD Radeon Vega 10 Graphics (RADV RAVEN)
Present mode: Mailbox
validation layer: Validation Error: [ VUID-VkBufferCreateInfo-size-00912 ] | MessageID = 0x7c54445e | vkCreateBuffer: parameter pCreateInfo->size is zero The Vulkan spec states: size must be greater than 0 (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkBufferCreateInfo-size-00912)
validation layer: Validation Error: [ VUID-VkMemoryAllocateInfo-allocationSize-07899 ] | MessageID = 0x5e299387 | vkAllocateMemory: allocationSize is 0. The Vulkan spec states: If the parameters define an export operation and the handle type is not VK_EXTERNAL_MEMORY_HANDLE_TYPE_ANDROID_HARDWARE_BUFFER_BIT_ANDROID, allocationSize must be greater than 0 (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-VkMemoryAllocateInfo-allocationSize-07899)
validation layer: Validation Error: [ UNASSIGNED-GeneralParameterError-RequiredParameter ] | MessageID = 0x99fb7dfd | vkBindBufferMemory: required parameter memory specified as VK_NULL_HANDLE
validation layer: Validation Error: [ VUID-vkBindBufferMemory-memoryOffset-01031 ] Object 0: VK_NULL_HANDLE, type = VK_OBJECT_TYPE_DEVICE_MEMORY; Object 1: handle = 0x8fd57200000000e5, type = VK_OBJECT_TYPE_BUFFER; | MessageID = 0xf77503bd | In vkBindBufferMemory(), attempting to bind VkDeviceMemory 0x0[] to VkBuffer 0x8fd57200000000e5[], memoryOffset=0x0 must be less than the memory allocation size 0x0. The Vulkan spec states: memoryOffset must be less than the size of memory (https://www.khronos.org/registry/vulkan/specs/1.3-extensions/html/vkspec.html#VUID-vkBindBufferMemory-memoryOffset-01031)
```

OSMaxwell, 2023-10-13:
[image]

Atraxus, 2023-10-15:
It does not crash for me though there is some weirdness going on with the rendering. I will create another issue for that problem. Would you mind testing again if it still crashes on your system @memoryleak47?

memoryleak47, 2023-10-16:
Doesn't crash anymore. Thanks!

closed 2023-10-16 by memoryleak47

OSMaxwell, 2023-10-19:
PS: This doesn't crash anymore because of e799f972c052be7313686d4b32fea52a6a8e1020

Resetting that to false, produces empty vertex list again. --> Empty buffer.

The new bool arg isn't nice in the Collision func call, however until the toroid dot prod. normal x ray is fixed, it remains.