hz-b/rayx #194: CMakeLists.txt file is missing in Extern/ImGuiFileDialog

treegardel, 2023-10-11:
**Describe the bug**
configuration for building rayx gives an error:

**To Reproduce**
1. get newest version via git: git pull --recurse-submodules
2. start clean build in visual code: cmake: "Clean Reconfigure All Projects"

**Expected behavior**
Configurations ends up with no errors

**Screenshots**
Instead it stops here:
`[cmake] -- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter `
`[cmake] CMake Error at Extern/CMakeLists.txt:6 (add_subdirectory):`
`[cmake]   The source directory`
`[cmake]`
`[cmake]     /home/usrename/RAY/RAY-X/Extern/ImGuiFileDialog`
`[cmake]`
`[cmake]   does not contain a CMakeLists.txt file.`
`[cmake]`
`[cmake]`
`[cmake] CMake Error at Extern/CMakeLists.txt:8 (target_include_directories):`
`[cmake]   Cannot specify include directories for target "ImGuiFileDialog" which is`
`[cmake]   not built by this project.`
`[cmake]`
`[cmake]`
`[cmake] -- Configuring incomplete, errors occurred!`
**Desktop (please complete the following information):**
 - Ubuntu 22.04

**Additional context**
CMakeLists.txt file is missing in my folder

OSMaxwell, 2023-10-11:
Maybe this is also necessary?..
git submodule update --init --recursive

memoryleak47, 2023-10-11:
> Additional context
CMakeLists.txt file is missing in my folder

Does `git status` also show this file as removed?
If yes, do `git checkout -- CMakeLists.txt` to get it back.

closed 2023-10-11 by treegardel