hz-b/rayx #197: Rendering issue toroid

Atraxus, 2023-10-15:
**Describe the bug**
When rendering the toroid beamline, the rays or the toroid are not placed correctly. Further not all rays seem to be renderd

**To Reproduce**
Steps to reproduce the behavior:
1. Open rayx-ui
2. Load this rml file 
[zippy.zip](https://github.com/hz-b/rayx/files/12909843/zippy.zip)
3. Go to the toroid optical element

**Expected behavior**
The rays should hit the object and no orange rays should just end without another bounce

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows

Atraxus, 2023-10-15:
@memoryleak47, do you know why some rays are reflected by the toroid (marked red) and some run out of bounces (marked pink). The rays that run out of bounces are (temporarily) rendered in white here
[image]

memoryleak47, 2023-10-16:
@Atraxus

> do you know why some rays are reflected by the toroid (marked red) and some run out of bounces (marked pink). The rays that run out of bounces are (temporarily) rendered in white here.

Yes, this is what I described in the meeting last week. Due to an imprecision in the Toroid collision, some rays repeatedly collide with the toroid until they are out of bounces.

This was fixed in ad5329d, the zip file contains rays from an older version.

Atraxus, 2023-10-18:
Fixed with 7105f4ec8ce9a792b6a33482f51c95229039ca03

closed 2023-10-18 by Atraxus