hz-b/rayx #198: benchmark.py doesn't work (on linux)

memoryleak47, 2023-10-16:
I've tried using bechmark.py and identified the following problems:
- it tries to use an executable named rayx.exe that doesn't exist on linux
- It cannot be called from the Scripts directory, instead you need to be in the repository root and call it as `python Scripts/benchmark.py`
- path_to_input looks for a directory named "scripts" and not "Scripts" (how did this ever work for anyone? is windows case insensitive?)
- what even is this output:
```
,Value
CPU model,x86_64
CPU cores,8
Total RAM,5.74GB
GPU model,No GPU found
Total GPU memory,N/A
```

It doesn't even show the time it took. (Should I run this within `time`?)

EnricoAhlers, 2023-10-23:
The instrumentor currently doesn't work at all on development and needs to be fixed first. Then we can fix the benchmark function for Linux.

Atraxus, 2023-10-24:
TODO: decide to do benchmarking during runtime and remove json functionality

EnricoAhlers, 2023-10-25:
Benchmarking works now on Ubuntu

EnricoAhlers, 2023-10-25:
I got very interesing results:
[image]

EnricoAhlers, 2023-10-25:
Needs further testing to see whether the OS or the compiler makes such a difference.

memoryleak47, 2023-10-31:
@EnricoAhlers just fixed the aforementioned issues. Thanks!

closed 2023-10-31 by memoryleak47