hz-b/rayx #232: Integrate RAYX into bluesky

EnricoAhlers, 2023-12-08:
First step: Providing a Python API for RAYX similar to https://github.com/hz-b/raypyng.

It would be good to add RAYX functionality directly to raypyng.
The issue: Ray-UI runs in interactive mode while RAYX currently is one-shot. We need to discuss whether we want an interactive mode CLI for RAYX too.

EnricoAhlers, 2024-04-09:
I want to have the possibility to tell Ray-X:

In The RAY-UI way:

which file to trace
define for which objects I want to export the rays
specify the names of the exported files. Or, at least, I need to prepend something to the filenames (normally a number,'n_')
the folder where I want the results to be saved.
Even Better:

I want to have the possibility to pass an RML that I have in memory to RAY-X, or pass the path of the rml file to load.
define for which objects I want to export the rays. Even better I would like to be able to specify some properties. Numbers of rays, absolute intensity (If this is known for the source), fwhm horizontal and vertical, bandwidth.
I want the option to save the results as files, specifying the folder and the filename, or I want the wrapper to just return the results, and then I can decide what to do with them.

Atraxus, 2024-09-18:
Blocked by #323

Atraxus, 2025-04-14:
There are some other requirements to rayx we need to work on before continuing this issue. We should look at some use-cases though, to figure out what is still missing. Simone offered to provide some examples.

TODO:

- [ ] List requirements in upcoming meeting
- [ ] Someone needs to look at the use-cases
- [ ] Figure out missing requirements
- [ ] Figure out what if we can already start working on a prototype