hz-b/rayx #238: Correctness of the Stokes vector calculation

memoryleak47, 2024-01-12:
It could be that the changes from Ray-UI to Ray-X make our calculation of stokes vectors invalid.
As Peter said, it might be that the calculation from Ray-UI only makes sense in a sequentially-traced setting.

We currently use the azimuthalAngle of OpticalElements to update the stokes vectors of our Ray.
The azimuthalAngle expresses a certain rotation "relative" to the previous OpticalElement.
And as our objects don't have a fixed order anymore, I suggest that this might not make much sense.

Generally, we should get rid of the azimuthalAngle as a model parameter! This is its last usage.

Important to note: stokes vectors checks are currently not done by the test suite.

First of all, we'd need to make sure that it works, or that it doesn't work. It should be covered by our test suite.

Potential pathes forward (assuming the current implementation is faulty):
- Peter is looking for an element-order independent approach using Muller-matrices
- He also mentioned that we could replace stokes vectors by the E-field directly, or other potential parameters
- We could accept that it doesn't work, and remove it for now

memoryleak47, 2024-02-16:
Our python experiments directly using the e-field are here: [e_field.zip](https://github.com/hz-b/rayx/files/14316133/e_field.zip).
The code is (imo) pretty well-documented, with references to wikipedia for the formulas & explanations.

memoryleak47, 2024-02-16:
Stokes Vector - Coordinate system thoughts:

The problem with the current stokes vector computation is that stokes vectors require to be expressed in a fixed coordinate system.
In particular, this is necessary for phi, which expresses the angle between the major semi-axis (of the polarization ellipsis) and the x-axis (of our fixed coordinate system!), see https://en.wikipedia.org/wiki/Stokes_parameters#Definitions.
In Ray-UI this coordinate system was the Beam-coordinate system.
We don't have that coordinate system anymore for rayx, hence we need to fix another coordinate system.

Also: one axis of this coordinate system needs to be the direction of the ray!

As one axis is already given by the direction of the ray, we only need one other orthogonal vector to complete a basis (as the third vector is just the cross product from the first two).

The question is, how do you decide on this second basis vector.
In my opinion, it can be chosen arbitrarily, as long as it is deterministic and orthogonal to the direction k.
Hence, one choice would be
```python
a = cross(k, [1, 0, 0])
b = cross(k, [0, 1, 0])

# c is not parallel to k, and non-zero
c = a + b

b1 = k
b2 = cross(k, c)
b3 = cross(b1, b2)
# b1, b2, b3 form a new basis in which the stokes vectors can be expressed
```

There are many choices that should work for this. The advantage of this concrete approach might be that it is a rather numerically stable I believe.

Peter argues that we need to store the coordinate system "within" the Ray. I, on the other hand, claim that the coordinate system can be inferred from the direction of the ray alone.

ichinii, 2024-07-30:
We decided to trace electric fields for efficiency calculation. closed.

closed 2024-07-30 by ichinii