hz-b/rayx #277: Fixing Ubuntu/Linux issues

Atraxus, 2024-04-24:
There are several issues with the current master branch and Linux. We need to make sure that it builds with clang++ and GCC as well as in debug and release. RAYX-UI has at least two bugs on Linux: NFD file dialog crashes on some systems and there are some RML files, which trigger the `[                xml.cpp:221] Vectors are not a basis.` error. This is strange for two reasons: this code should not be included in release builds and the RML works on other systems (at least on Windows). The xml.cpp has not changed in a long time, which makes this even stranger.

Further, considering the last failure of the CI (ubuntu and ubuntuClang) the CI scripts might need updates. Maybe the GitHub runners have undergone some changes

Concerning the file dialog crashes we need to contact the developers of the NFD-extended library and ask them how to debug this as it does not seem straight forward.

Atraxus, 2024-04-26:
On my Ubuntu system I could fix some issues by:
- using header guards instead of `#pragma once` in H5Writer.h, since `#pragma once` was not working correctly with clang++
- I needed to do `unset GTK_PATH` so I could launch the rayx-ui on nvidia gprahics

EnricoAhlers, 2024-05-06:
"Vectors are not a basis." is fixed on the branch "ubuntu-fixes". The issue was the system language. It's now always set to C.

Atraxus, 2024-06-11:
I changed the file dialog library on #290. It still needs to be tested

closed 2024-07-09 by Atraxus