hz-b/rayx #368: RayUI RayX footprint differences

julixH, 2025-05-26:
The footprints generated with RayUI and RayX differ for several elements. See info.txt

[rayXrayUI.zip](https://github.com/user-attachments/files/20442315/rayXrayUI.zip)

julixH, 2025-06-02:
I replaced these values in the rml 

Ellipsoid: 
`<param id="misalignmentCoordinateSystem" comment="Mirror (standard)" enabled="T">1</param>`

Grating:
`<param id="gratingEfficiency" enabled="T">1</param>` 

and for all elements: 
` <param id="reflectivityType" comment="Derived by Material" enabled="T">0</param>

and now rayX and rayUI produce the same results.

treegardel, 2025-06-03:
How are the hist projections are calculated from the footprints in the comparison of the Ellipsoid and Grating? (https://github.com/hz-b/rayx/issues/368#issuecomment-2930406495) In the case of "hist projection x" (which should called hist projection y, or?), the scale of the x-axes does not fit to the footprint. Each bin has a certain x-value and a certain width, are these definitions for both cases the same? Especially in the footprint of the ellipsoid, I cannot see the relatively large offset in the y-direction that can be seen in the projection.

julixH, 2025-06-03:
For the metrix beamline elements, I guess the world coordinates and x/y/z direction to not match the other parameters. I updated the value in distancePreceding to worldPosition.z. But which parameters are responsible for the x/y/z direction?

julixH, 2025-06-03:
I changed ` <param id="misalignmentCoordinateSystem" comment="Mirror (standard)" enabled="T">1</param>' in the complete metrix beamline and now the results look the same for all sampled configurations specified in the paper [Offset Finding of Beamline Parameters](https://arxiv.org/abs/2503.17396). 
So I think this issue is not relevant anymore :)

treegardel, 2025-06-03:
In the case of the grating there is still a difference in the maximal hight in the hist projection y. Looking at the raw footprint data it seems that in the case of RAYX there are few rays which are further away from zero (max y = -0.175) than in the case of RAY-UI (max y = -0.13). In case your bin size depends on this extrema and you choose in both cases the same number of bins, this could explain the height difference. As in the case of RAY-UI the bin width would be smaller.

treegardel, 2025-06-16:
Last open question in this issue: is RAYX throwing away rays (using Monte Carlo) for tracing the reflection or is RAYX multiplying the reflectivity to the E-Field?